Xiangxu-0103/FRNet #3: About frustum mix

hyalvin, 2023-12-27:
Hi, thanks for the nice work~

I'm a little bit confused about the difference between `frustum_vertical_mix_transform` in your code and LaserMix? Also, what is the difference between `frustum_horizental_mix_transform` and the `ablation study on laser beam partitions` in LaserMix (showed in Table4)?

Xiangxu-0103, 2023-12-27:
Hi @hyalvin , thanks for your interest in our work.

Our `FrustumMix` is a modified version of `LaserMix` and `PolarMix`, which mixed two scans based on the frustum region, while `LaserMix` and `PolarMix` mixed two scans at point level. (`frustum_vertical_mix` corresponds to `LaserMix`, and `frustum_horizontal_mix` corresponds to `PolarMix`).

Point-level mixing strategy may break the structure in the frustum region while frustum-level can preserve the frustum structure, which performs better in our frustum-based methods (See Fig. 5 (b) and (c) for compared results).

Hope these replies can help you.

hyalvin, 2023-12-28:
Thanks for your kind reply!

As you mentioned above,  `frustum_vertical_mix corresponds to LaserMix`. From my naively perspective, I think it is equivalent to split the area using coordinates under range-view representation and using yaw/pitch under point representation. Cause the area range and `num_areas` is decided randomly, so both of them could break the point structure possibly.

Just curious about the above standpoint, hope for your reply, thanks!

Xiangxu-0103, 2023-12-29:
Yeah, you are right. Different from `LaserMix`, the coordinates are calculated as continuous values, while `FrustumMix` discretizes them based on the frustum region. 

I mean that we want to not break the structure of points in the frustum region instead of the whole scan scene.

hyalvin, 2023-12-29:
Thanks for the detailed explanation, I have got the point.
Just can't wait to see this nice work being accepted!

closed 2024-02-19 by Xiangxu-0103