Xiangxu-0103/FRNet #5: About your paper.

hyalvin, 2024-01-09:
[image]
Here, if following the representation in your code as:
[image]
should `f_up` be `|f_down|`?

Xiangxu-0103, 2024-01-11:
Sorry for the late reply. Thanks for your pointing out the problem. It seems an error in the paper, which is also presented in RangeNet++. We will modify the equation in the next version.

hyalvin, 2024-01-11:
> Sorry for the late reply. Thanks for your pointing out the problem. It seems an error in the paper, which is also presented in RangeNet++. We will modify the equation in the next version.

One more potential error,
[image]
Here, I guess f_ik means the feature of the k-th point in the i-th frustum region.

Xiangxu-0103, 2024-01-11:
@hyalvin Thanks again for your correction.

closed 2024-03-23 by Xiangxu-0103