Xiangxu-0103/FRNet #7: Some questions

Wansit99, 2024-01-14:
Thank you very much for your excellent work. May I inquire about how to resume training if it was interrupted midway? Additionally, could you please let me know where I can find the code for calculating the Frustum? Thank you!

Xiangxu-0103, 2024-01-14:
Thanks for your interest in our work.

1. If you want to resume training. You can add `--resume`  argument when launching the training script with the following two methods:
   - `--resume`: it will automatically load the latest checkpoint.
   - `--resume ${path/to/the/checkpoint}`: it will load the checkpoint you specified.
 
2. For the `frustum-level loss`, you can refer to the code [here](https://github.com/Xiangxu-0103/FRNet/blob/master/frnet/models/decode_heads/frustum_head.py#L65), and the pseudo frustum label generated [here](https://github.com/Xiangxu-0103/FRNet/blob/master/frnet/models/data_preprocessors/data_preprocessor.py#L110).

Wansit99, 2024-01-15:
I have another question. In your opinion, what do you think is the bottleneck for training neural networks? It takes me approximately three days using a single RTX 3090

Xiangxu-0103, 2024-01-15:
We trained FRNet with 4 RTX 2080Ti GPUs and it will take around 25 hours due to the design of `frustum-level` supervision. It will be much faster w/o this module.

If you use `RTX 3090`, you may try to set a larger batch size.

Wansit99, 2024-03-16:
> We trained FRNet with 4 RTX 2080Ti GPUs and it will take around 25 hours due to the design of `frustum-level` supervision. It will be much faster w/o this module.
> 
> If you use `RTX 3090`, you may try to set a larger batch size.

Hello, I've looked through the code for the loss function, but which one is the frustum-level loss? I feel like it's the point-level loss."

Wansit99, 2024-03-16:
> We trained FRNet with 4 RTX 2080Ti GPUs and it will take around 25 hours due to the design of `frustum-level` supervision. It will be much faster w/o this module.
> 
> If you use `RTX 3090`, you may try to set a larger batch size.

Hello, I've reviewed the configuration file and code related to the loss. I'm curious about how the network would predict labels if we remove the frustum-level loss. All loss functions in the configuration file are located within the FrustumHead

Wansit99, 2024-03-16:
> We trained FRNet with 4 RTX 2080Ti GPUs and it will take around 25 hours due to the design of `frustum-level` supervision. It will be much faster w/o this module.
> 
> If you use `RTX 3090`, you may try to set a larger batch size.

Hello, I removed all auxiliary heads in the configuration file, but the speed did not significantly improve. Could there be an issue with the way I made the modifications?

Xiangxu-0103, 2024-03-16:
Hi, which speed do you mean? Training or inference?

Auxiliary heads are only used during training, and not used during inference. Removing the auxiliary head will speed up training (with lower performance), and not influence the inference speed.

Moreover, we are going to update our code to make performance more stable in the updated version.

Wansit99, 2024-03-17:
> Hi, which speed do you mean? Training or inference?
>
> Auxiliary heads are only used during training, and not used during inference. Removing the auxiliary head will speed up training (with lower [...]

Thank you for your response. What I meant by training time was the issue of training time being excessively long. In your previous response, you mentioned that removing supervision at the frustum level would significantly improve speed. By the way, may I ask when the code will be updated? Looking forward to your new work. Could you please help me translate this into English?

Xiangxu-0103, 2024-03-18:
Thanks for your attention. We plan to update our report and the code this month.

Wansit99, 2024-03-19:
> Thanks for your attention. We plan to update our report and the code this month.

Looking forward to your new job!