Xiangxu-0103/FRNet #8: test time augmentation

Wansit99, 2024-02-13:
Happy New Year! Did you use test time augmentation technique during valing and testing?

Xiangxu-0103, 2024-02-13:
The benchmark results reported in our paper are evaluated with TTA (with 4 votes).

If you want to evaluate with TTA, you can add `--tta` while running the testing script.

Wansit99, 2024-02-14:
Thank you for such a quick response. Happy New Year!

> The benchmark results reported in our paper are evaluated with TTA (with 4 votes).
> 
> If you want to evaluate with TTA, you can add `--tta` while running the testing script.

Thank you for such a quick response. Happy New Year!

Wansit99, 2024-02-14:
> The benchmark results reported in our paper are evaluated with TTA (with 4 votes).
> 
> If you want to evaluate with TTA, you can add `--tta` while running the testing script.

Thank you for such a quick response. Happy New Year!

Wansit99, 2024-02-14:
I also want to ask, when performing the test, do I need to fine-tune it on the validation set? I got 67.6 on the validation set, but only 68.5 on the test set when using TTA.

Wansit99, 2024-02-14:
I am currently experiencing some difficulties while trying to replicate the paper. When I set FrustumMix to 512, I only achieve an accuracy of around 66.7% on the validation set, which is significantly lower than the reported numbers in the paper. My experimental setup is identical to the one described in the paper, except that I am using two NVIDIA RTX 3090 GPUs with a batch size of 3 per GPU. The accuracy of 67.6% was obtained when I increased FrustumMix to 1024. Do you have any insights into what might be causing this discrepancy in accuracy? I would greatly appreciate your help in identifying potential issues or suggestions for further investigation. Thank you.

Xiangxu-0103, 2024-02-17:
> I also want to ask, when performing the test, do I need to fine-tune it on the validation set? I got 67.6 on the validation set, but only 68.5 on the test set when using TTA.

Sorry for the late reply. For the test set, we train on the `train-val` set and use some other tricks as many works do.

Xiangxu-0103, 2024-02-17:
> I am currently experiencing some difficulties while trying to replicate the paper. When I set FrustumMix to 512, I only achieve an accuracy of around 66.7% on the validation set, which is [...]

Whether the `66.7` tested with TTA? The performance shown in the training log is tested without TTA augmentation. You may also try to set larger learning rate if you use larger batch size than we provided.

Wansit99, 2024-02-17:
> Whether the `66.7` tested with TTA? The performance shown in the training log is tested without TTA augmentation. You may also try to set larger learning rate if you use larger batch size than we [...]

Thank you for the reply. I have another question I'd like to ask for your advice. In the paper, it's mentioned that using a 2080Ti allows for a batch size of 2, but with my 3090, I can only set the batch size to 3. Setting it to 4 causes an out-of-memory error. What could be the reason for this? Why does adding one more batch size consume so much more memory? Are there any optimizations that can be made?

Wansit99, 2024-02-17:
> > I also want to ask, when performing the test, do I need to fine-tune it on the validation set? I got 67.6 on the validation set, but only 68.5 on the test set when using TTA.
> 
> Sorry for the late reply. For the test set, we train on the `train-val` set and use some other tricks as many works do.

Thanks！

Wansit99, 2024-02-17:
> Whether the `66.7` tested with TTA? The performance shown in the training log is tested without TTA augmentation. You may also try to set larger learning rate if you use larger batch size than we [...]

I also wanted to ask, if I set the batch size to 3 and use 2 GPUs, should I reduce my learning rate? The paper used a batch size of 2 with 4 GPUs

Wansit99, 2024-02-17:
> Whether the `66.7` tested with TTA? The performance shown in the training log is tested without TTA augmentation. You may also try to set larger learning rate if you use larger batch size than we [...]

66.7 is obtained from the training log without using TTA. Currently, with a batch size of 3 on 2 GPUs, the highest validation accuracy achieved is 67.46 without changing the learning rate

Xiangxu-0103, 2024-02-19:
> Thank you for the reply. I have another question I'd like to ask for your advice. In the paper, it's mentioned that using a 2080Ti allows for a batch size of 2, but with my 3090, I can only set the [...]

It seems `naiveSyncBatchNorm` implemented in mmdet3d may require more memory compared to common BN in `torch` (which achieves better performance than `SyncBN`).

Wansit99, 2024-02-21:
> 1. It seems `naiveSyncBatchNorm` implemented in mmdet3d requires more memory compared to common `BN` in `pytorch` (which achieves better performance than `SyncBN`).
> 2. It is suggested to adjust learning rate if your setting is different from ours, you can use `--auto-scale-lr` when launching the training script.
> 3. From my perspective, the result seems OK as we also report the best mIoU during the training procedure. A small range of floating comes from environments, such as `random seed`, `CUDA version`, and others.

Thanks!

Lizhijian-123, 2024-03-20:
> > I also want to ask, when performing the test, do I need to fine-tune it on the validation set? I got 67.6 on the validation set, but only 68.5 on the test set when using TTA.
> 
> Sorry for the late reply. For the test set, we train on the `train-val` set and use some other tricks as many works do.

hello,can i ask what is the tricks?

Lizhijian-123, 2024-03-20:
> The benchmark results reported in our paper are evaluated with TTA (with 4 votes).
> 
> If you want to evaluate with TTA, you can add `--tta` while running the testing script.
Hello, I would like to ask why tta does not use a larger num_vote? I see some codes use num_vote=12.

Lizhijian-123, 2024-03-20:
> I also want to ask, when performing the test, do I need to fine-tune it on the validation set? I got 67.6 on the validation set, but only 68.5 on the test set when using TTA.

May I ask how many val miou and test miou you had in semanticKITTI before using TTA? How much after tta? I saw that your test miou in semanticKITTI was only about 68.7 before, and it reached 73.3 in the paper. Have you reproduced this result? What is your current test miou? Do you have any tricks?Thanks

closed 2024-03-23 by Xiangxu-0103