Xiangxu-0103/FRNet #9: KeyError: 'pts_semantic_mask_path' while testing

tt-232, 2024-02-24:
Hi, thank you so much for your awesome work!

I am testing with your pre-trained model on nuscenes test set and am facing this error while launching the test program. Could you please give me a hint which part could be wrong?

Thank you in advance!

```
Traceback (most recent call last):
  File "test.py", line 146, in <module>
    main()
  File "test.py", line 142, in main
    runner.test()
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmengine/runner/runner.py", line 1823, in test
    metrics = self.test_loop.run()  # type: ignore
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmengine/runner/loops.py", line 442, in run
    for idx, data_batch in enumerate(self.dataloader):
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 517, in __next__
    data = self._next_data()
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1199, in _next_data
    return self._process_data(data)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1225, in _process_data
    data.reraise()
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/_utils.py", line 429, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/miniconda3/envs/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 403, in __getitem__
    data = self.prepare_data(idx)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmdet3d/datasets/seg3d_dataset.py", line 305, in prepare_data
    return super().prepare_data(idx)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 793, in prepare_data
    return self.pipeline(data_info)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 60, in __call__
    data = t(data)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmcv/transforms/base.py", line 12, in __call__
    return self.transform(results)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmdet3d/datasets/transforms/loading.py", line 1070, in transform
    results = self._load_semantic_seg_3d(results)
  File "/root/miniconda3/envs/lib/python3.8/site-packages/mmdet3d/datasets/transforms/loading.py", line 955, in _load_semantic_seg_3d
    pts_semantic_mask_path = results['pts_semantic_mask_path']
KeyError: 'pts_semantic_mask_path'
```

Xiangxu-0103, 2024-02-25:
Hi @tt-232 , `test set` does not have annotation files. If you want to evaluate on test set, you should save the prediction results and upload them to evaluation proto.

Also, you need to comment `LoadAnnotations3D` and `PointSegClassMapping` in `test_pipeline`.

tt-232, 2024-03-03:
Hi, @Xiangxu-0103, thank you so much for your reply and your help! I got the results.

I would like to ask if you provide an approach to save the prediction results in the submission format.

Xiangxu-0103, 2024-03-04:
Sorry for that we do not have plans to support this feature in this repo recently. If you want to save prediction results, you can customize a new `Metric` method instead of `SegMetric` in the config file.

Moreover, we will consider supporting this feature in `mmdet3d` repo in the future.

tt-232, 2024-03-04:
> Sorry for that we do not have plans to support this feature in this repo recently. If you want to save prediction results, you can customize a new `Metric` method instead of `SegMetric` in the config file.
> 
> Moreover, we will consider supporting this feature in `mmdet3d` repo in the future.

Thank you so much for the support!

closed 2024-03-04 by tt-232