Xiangxu-0103/FRNet #10: About infer

Wansit99, 2024-03-15:
May I ask how I can use FRNet for inference to obtain visualized results? Currently, I can get the mIoU, but I also want to obtain the corresponding point cloud inference results. Thank you!

Xiangxu-0103, 2024-03-16:
For visualization, you may refer to [mmdet3d](https://github.com/open-mmlab/mmdetection3d) for details.

Wansit99, 2024-03-16:
> For visualization, you may refer to [mmdet3d](https://github.com/open-mmlab/mmdetection3d) for details.

Thanks！

Lizhijian-123, 2024-03-20:
> May I ask how I can use FRNet for inference to obtain visualized results? Currently, I can get the mIoU, but I also want to obtain the corresponding point cloud inference results. Thank you!

May I ask how many val miou and test miou you had in semanticKITTI before using TTA? How much after tta? I saw that your test miou in semanticKITTI was only about 68.7 before, and it reached 73.3 in the paper. Have you reproduced this result? What is your current test miou? Do you have any tricks?Thanks

closed 2024-03-23 by Xiangxu-0103