Xiangxu-0103/FRNet #11: Label-Efficient LiDAR Segmentation

matildecc, 2024-03-15:
Thank you for sharing your work and congratulations on the impressive results. I would like to inquire about the semi-supervised task in more detail. In the paper, you mentioned 'following the lasermix paradigm'. Does that mean you followed the same training pipeline and configurations, but only changed the backbone to FRNet and the type of augmentation to frustumMix instead of lasermix? Do you have any code related to this experiment? Did you utilize the lasermix repository and simply modify the backbone and augmentation type?

Xiangxu-0103, 2024-03-16:
For semi-supervised, you can refer to [LaserMix](https://github.com/ldkong1205/LaserMix) for details.

We use FRNet as the backbone and replace the `LaserMix` with `FrustumMix` to mix labeled and unlabeled data here.

closed 2024-03-23 by Xiangxu-0103

matildecc, 2024-05-23:
What about the losses? Did you use the supervised loss proposed in your paper or the one used in the LaserMix paper?
Additionally, what about the RangeInterpolation augmentation? Did you also apply it in this study?

If you could provide more details (namely your configurations or any code) about how you performed this comparative study, I would greatly appreciate it.