Xiangxu-0103/FRNet #12: About Test 11-21 sequences

Lizhijian-123, 2024-03-20:
Hello, I want to run the 11-21 sequence test set of semanticKITTI and then submit it to codalab. I used test.py and added tta to run and found that the 8th sequence of the validation set was run. Can you teach me how I can change the code to run 11-21 sequences? Including using tta. At the same time, I would like to ask what tricks  you used when submitting to codalab? I reproduced several codes and submitted them to codalab, but they were unable to achieve the results of their papers. I don’t know what tricks were used.

Xiangxu-0103, 2024-03-23:
For the `test set`, you need to 1) modify the `ann_file` in `test_dataloader` to `semantickitti_infos_test.pkl`; 2) uncomment `LoadAnnotations3D` and `PointSegClassMapping` in `test_pipeline` and `tta_pipeline`; 3) custom a new evaluator to save pred files and replace `test_evaluator` with your customized evaluator.

For submitting, we train on the `train-val` set and test with TTA, model ensembing, etc.

Xavier-wa, 2024-03-24:
> test_pipeline

hello，thanks for your opensoure,Here is my file: (Are there any error? I can run the infer)
```
# For SemanticKitti we usually do 19-class segmentation.
# For labels_map we follow the uniform format of MMDetection & MMSegmentation
# i.e. we consider the unlabeled class as the last one, which is different
# from the original implementation of some methods e.g. Cylinder3D.
dataset_type = 'SemanticKittiDataset'
data_root = 'data/semantickitti/'
class_names = [
    'car', 'bicycle', 'motorcycle', 'truck', 'bus', 'person', 'bicyclist',
    'motorcyclist', 'road', 'parking', 'sidewalk', 'other-ground', 'building',
    'fence', 'vegetation', 'trunck', 'terrian', 'pole', 'traffic-sign'
]
labels_map = {
    0: 19,  # "unlabeled"
    1: 19,  # "outlier" mapped to "unlabeled" --------------mapped
    10: 0,  # "car"
    11: 1,  # "bicycle"
    13: 4,  # "bus" mapped to "other-vehicle" --------------mapped
    15: 2,  # "motorcycle"
    16: 4,  # "on-rails" mapped to "other-vehicle" ---------mapped
    18: 3,  # "truck"
    20: 4,  # "other-vehicle"
    30: 5,  # "person"
    31: 6,  # "bicyclist"
    32: 7,  # "motorcyclist"
    40: 8,  # "road"
    44: 9,  # "parking"
    48: 10,  # "sidewalk"
    49: 11,  # "other-ground"
    50: 12,  # "building"
    51: 13,  # "fence"
    52: 19,  # "other-structure" mapped to "unlabeled" ------mapped
    60: 8,  # "lane-marking" to "road" ---------------------mapped
    70: 14,  # "vegetation"
    71: 15,  # "trunk"
    72: 16,  # "terrain"
    80: 17,  # "pole"
    81: 18,  # "traffic-sign"
    99: 19,  # "other-object" to "unlabeled" ----------------mapped
    252: 0,  # "moving-car" to "car" ------------------------mapped
    253: 6,  # "moving-bicyclist" to "bicyclist" ------------mapped
    254: 5,  # "moving-person" to "person" ------------------mapped
    255: 7,  # "moving-motorcyclist" to "motorcyclist" ------mapped
    256: 4,  # "moving-on-rails" mapped to "other-vehic------mapped
    257: 4,  # "moving-bus" mapped to "other-vehicle" -------mapped
    258: 3,  # "moving-truck" to "truck" --------------------mapped
    259: 4  # "moving-other"-vehicle to "other-vehicle"-----mapped
}
metainfo = dict(
    classes=class_names, seg_label_mapping=labels_map, max_label=259)
input_modality = dict(use_lidar=True, use_camera=False)

# Example to use different file client
# Method 1: simply set the data root and let the file I/O module
# automatically infer from prefix (not support LMDB and Memcache yet)

# data_root = 's3://openmmlab/datasets/detection3d/semantickitti/'

# Method 2: Use backend_args, file_client_args in versions before 1.1.0
# backend_args = dict(
#     backend='petrel',
#     path_mapping=dict({
#         './data/': 's3://openmmlab/datasets/detection3d/',
#          'data/': 's3://openmmlab/datasets/detection3d/'
#      }))
backend_args = None

[... 188 lines cut ...]
    drop_last=False,
    sampler=dict(type='DefaultSampler', shuffle=False),
    dataset=dict(
        type=dataset_type,
        data_root=data_root,
        ann_file='semantickitti_infos_test.pkl',
        pipeline=test_pipeline,
        metainfo=metainfo,
        modality=input_modality,
        ignore_index=19,
        test_mode=True,
        backend_args=backend_args))
test_dataloader = val_dataloader

val_evaluator = dict(type='SegMetric')
test_evaluator = dict(type='SegMetric',submission_prefix="./result/")

vis_backends = [dict(type='LocalVisBackend')]
visualizer = dict(
    type='Det3DLocalVisualizer', vis_backends=vis_backends, name='visualizer')

tta_model = dict(type='Seg3DTTAModel')

``` 
I am a beginner in mm3d. Could you tell me why?Thanks.

Xiangxu-0103, 2024-03-25:
Hi @Xavier-wa. For the `test set`, you should comment `LoadAnnotations3D` and `PointSegClassMapping` in `test_pipeline` and `tta_pipeline` as no annotations are available for the `test set`.

Xavier-wa, 2024-03-25:
> Hi @Xavier-wa. For the `test set`, you should comment `LoadAnnotations3D` and `PointSegClassMapping` in `test_pipeline` and `tta_pipeline` as no annotations are available for the `test set`.

Thank your for your help. I create a new test_evaluator and it works! thx!