Xiangxu-0103/FRNet #16: Testing issues：OSError: Caught OSError in DataLoader worker process 0.   And OSError: [Errno 5] Input/output error

xiaosa269, 2024-03-29:
__Thank you very much for your contribution. I encountered an interruption while testing SemanticKITTI using your Pre-Trained Checkpoints with the command:
python test.py /home/xhy/code/FRNet-master/configs/frnet/frnet-semantickitti_seg.py /home/xhy/code/FRNet-master/frnet-semantickitti_seg.pth

The generated error is as follows：

 python test.py /home/xhy/code/FRNet-master/configs/frnet/frnet-semantickitti_seg.py /home/xhy/code/FRNet-master/frnet-semantickitti_seg.pth
03/29 13:59:11 - mmengine - INFO - 
------------------------------------------------------------
System environment:
    sys.platform: linux
    Python: 3.8.0 (default, Nov  6 2019, 21:49:08) [GCC 7.3.0]
    CUDA available: True
    numpy_random_seed: 746820188
    GPU 0,1: NVIDIA GeForce RTX 3060
    CUDA_HOME: /usr/local/cuda
    NVCC: Cuda compilation tools, release 11.3, V11.3.58
    GCC: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
    PyTorch: 1.8.1+cu111
    PyTorch compiling details: PyTorch built with:
  - GCC 7.3
  - C++ Version: 201402
  - Intel(R) Math Kernel Library Version 2020.0.0 Product Build 20191122 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v1.7.0 (Git Hash 7aed236906b1f7a05c0917e5257a1af05e9ff683)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 11.1
  - NVCC architecture flags: -gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
  - CuDNN 8.0.5
  - Magma 2.5.2
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=11.1, CUDNN_VERSION=8.0.5, CXX_COMPILER=/opt/rh/devtoolset-7/root/usr/bin/c++, CXX_FLAGS= -Wno-deprecated -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -fopenmp -DNDEBUG -DUSE_KINETO -DUSE_FBGEMM -DUSE_QNNPACK -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -O2 -fPIC -Wno-narrowing -Wall -Wextra -Werror=return-type -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-unused-result -Wno-unused-local-typedefs -Wno-strict-overflow -Wno-strict-aliasing -Wno-error=deprecated-declarations -Wno-stringop-overflow -Wno-psabi -Wno-error=pedantic -Wno-error=redundant-decls -Wno-error=old-style-cast -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, PERF_WITH_AVX512=1, TORCH_VERSION=1.8.1, USE_CUDA=ON, USE_CUDNN=ON, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=ON, USE_NNPACK=ON, USE_OPENMP=ON, 

    TorchVision: 0.9.1+cu111
    OpenCV: 4.9.0
    MMEngine: 0.9.0

Runtime environment:
    cudnn_benchmark: False
    mp_cfg: {'mp_start_method': 'fork', 'opencv_num_threads': 0}
    dist_cfg: {'backend': 'nccl'}
    seed: 746820188
    Distributed launcher: none
    Distributed training: False
    GPU number: 1
------------------------------------------------------------

03/29 13:59:13 - mmengine - INFO - Config:
auto_scale_lr = dict(base_batch_size=4, enable=False)
backend_args = None
class_names = [
    'car',
    'bicycle',
    'motorcycle',
    'truck',
    'bus',
    'person',
    'bicyclist',
    'motorcyclist',
    'road',
    'parking',
    'sidewalk',
    'other-ground',
    'building',
    'fence',
    'vegetation',
    'trunck',
    'terrian',
    'pole',
    'traffic-sign',
]
[... 1193 lines cut ...]
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 403, in __getitem__
    data = self.prepare_data(idx)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmdet3d/datasets/seg3d_dataset.py", line 305, in prepare_data
    return super().prepare_data(idx)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 793, in prepare_data
    return self.pipeline(data_info)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmengine/dataset/base_dataset.py", line 60, in __call__
    data = t(data)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmcv/transforms/base.py", line 12, in __call__
    return self.transform(results)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmdet3d/datasets/transforms/loading.py", line 646, in transform
    points = self._load_points(pts_file_path)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmdet3d/datasets/transforms/loading.py", line 622, in _load_points
    pts_bytes = get(pts_filename, backend_args=self.backend_args)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmengine/fileio/io.py", line 181, in get
    return backend.get(filepath)
  File "/home/xhy/miniconda3/envs/frnet/lib/python3.8/site-packages/mmengine/fileio/backends/local_backend.py", line 34, in get
    value = f.read()
OSError: [Errno 5] Input/output error

Could you please advise on how to resolve this issue? Your help is greatly appreciated.

Xiangxu-0103, 2024-03-30:
Hi @xiaosa269 , thanks for your interest in our work. It seems that you should check your local environment.

Following situations will cause this problems:

- Corruption of your dataset.
- Local dist.
- Local memory is full.
- and others.

xiaosa269, 2024-03-30:
Hi@Xiangxu-0103 ,Thank you for your answer. I will search for the above issue and try again。

xiaosa269, 2024-03-30:
Thank you very much for your guidance and assistance. I have successfully executed it. Wish you smooth sailing on your research journey!

closed 2024-03-30 by xiaosa269