Xiangxu-0103/FRNet #17: About the result of test

longsijie, 2024-04-07:
Hi @Xiangxu-0103 Great work! I'm a beginner,and I have some questions about the result. I downloaded the checkpoint provided in the repo, and used this model to test. "python test.py ${CONFIG_FILE} ${CHECKPOINT_FILE} --work-dir xxx"
But finally I got  67.55% miou, which is far from the miou 73.3% mentioned in the article.
[image]
I don't know where the problem is. I hope you could tell me and look forward to your reply! thx!

Xiangxu-0103, 2024-04-07:
Hi @longsijie, thanks for your interest in our work.

The default `test_script` is conducted on the `val` set without TTA. If you want to evaluate under `test` set, you can follow issue #13 for details.

longsijie, 2024-04-07:
@Xiangxu-0103  Thank you for your reply！ But sorry I'm still a little confused. Can I achieve 73.3% miou by using the provided checkpoint and add -- TTA when test? Or I have to train a model by myself?

Xiangxu-0103, 2024-04-07:
1. Checkpoints provided in the repo are trained only under `train` set, which are not used to reproduce our results of `test` set.
2. If you want to evaluate on the `test` set, we suggest training on `train-val` set with model ensembing, TTA, etc.
3. For saving predict files, you need to custom `evalutor` instead of `SegMetric`, in which you should remap the predicted label to original label id and save the file.
4. For visualization, we suggest saving the prediction file and visualizing online, with Open3D (if supported) or matplotlib (for BEV or range view).

longsijie, 2024-04-08:
Thank you for your patient reply. I'll try it. Thx!

zhoujixiang, 2024-04-08:
How can I save the prediction file? Looking forward to a reply

zhoujixiang, 2024-04-08:
I have just started working in this field. How can I save prediction files？Could you provide a detailed solution? Looking forward to receiving your reply.

Xiangxu-0103, 2024-04-10:
For the saving prediction file, please kindly refer to #12.

For more advanced usage about customizing modules, you can refer to [mmengine tutorial](https://mmengine.readthedocs.io/en/latest/), [mmdet3d tutorial](https://mmdetection3d.readthedocs.io/en/latest/) for details.

closed 2024-04-14 by Xiangxu-0103

Hayleycat, 2025-02-26:
> 1. Checkpoints provided in the repo are trained only under `train` set, which are not used to reproduce our results of `test` set.
> 2. If you want to evaluate on the `test` set, we suggest training [...]

Hello, thank you for your excellent work. May I ask how to train on the train-val set and how to modify the code. I tried to increase the validation set, but found an error.

Xiangxu-0103, 2025-02-26:
@Hayleycat Thanks for your interest in our work. For training on the `train-val` set, you may follow the following steps:

1. Use our [script](https://github.com/Xiangxu-0103/FRNet/blob/master/tools/create_semantickitti.py) to generate corresponding `pkl` files, which should include `semantickitti_infos_trainval.pkl`.
2.  Modify the `ann_file` in [config file](https://github.com/Xiangxu-0103/FRNet/blob/master/configs/_base_/datasets/semantickitti_seg.py#L243) to `semantickitti_infos_trainval.pkl`.
3. Adjust the training iters in [config file](https://github.com/Xiangxu-0103/FRNet/blob/master/configs/_base_/schedules/onecycle-50k.py) based on your settings.

zhangsisi-git, 2025-03-05:
Hello, may I ask if you can provide a new evaluator created during testing? After debugging with the evaluator provided in question 13 and uploading codalab, the accuracy obtained was only 0.84.