Xiangxu-0103/FRNet #18: freeze model

Wansit99, 2024-04-10:
Hello, could you please advise on how to freeze network parameters? For example, if I only want to train `frner_head`.

Xiangxu-0103, 2024-04-10:
You can freeze some modules [here](https://github.com/Xiangxu-0103/FRNet/blob/master/frnet/models/segmentors/frnet.py#L57) by adding the following code (taking backbone as an example):

```python
self.backbone.eval()
for param in self.backbone.parameters():
    param.requires_grad = False
```

Hope the reply can help you.

Wansit99, 2024-04-10:
> You can freeze some modules [here](https://github.com/Xiangxu-0103/FRNet/blob/master/frnet/models/segmentors/frnet.py#L57) by adding the following code (taking backbone as an example):
>
> ```python [...]

Thanks！

closed 2024-04-14 by Xiangxu-0103