Xiangxu-0103/FRNet #19: param and fps

wp664, 2024-04-24:
Hello, could you please tell me how to obtain param and fps?

Xiangxu-0103, 2024-04-25:
Hi @wp664, for FPS, you can follow this [script](https://github.com/open-mmlab/mmdetection3d/blob/main/tools/analysis_tools/benchmark.py) provided in `mmdet3d`. While for param, you can refer to the [parameter_count_table](https://github.com/open-mmlab/mmengine/blob/main/mmengine/analysis/complexity_analysis.py#L295) implemented in `mmengine`.

wp664, 2024-04-25:
@Xiangxu-0103, Thanks.

wp664, 2024-04-25:
I ran the following command and got around 18 fps on the 3090, which is quite different from the 29.1 mentioned in the paper. Do I need to do anything else? Thank you for your help.
python mmdetection3d/tools/analysis_tools/benchmark.py configs/frnet/frnet-semantickitti_seg.py frnet-semantickitti_seg.pth

Xiangxu-0103, 2024-04-26:
Hi, the FPS is calculated under `fp16` mode.

closed 2024-05-15 by wp664