Xiangxu-0103/FRNet #21: About update

Wansit99, 2024-04-26:
I'm so glad you've updated your nice work. Could you please provide details on what specific updates have been made to enhance stability?Thanks!

Wansit99, 2024-04-26:
And when I update this files, I meet some problems

  File "/opt/data/private/whu_wsy/seg/FRNet-master-init/frnet/models/segmentors/frnet.py", line 83, in extract_feat
    voxel_dict = self.voxel_encoder(voxel_dict)
  File "/opt/conda/envs/frnet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/opt/data/private/whu_wsy/seg/FRNet-master-init/frnet/models/voxel_encoders/frustum_encoder.py", line 106, in forward
    voxel_feats = torch_scatter.scatter_max(
RuntimeError: The following operation failed in the TorchScript interpreter.
Traceback of TorchScript (most recent call last):
  File "/opt/conda/envs/frnet/lib/python3.8/site-packages/torch_scatter/scatter.py", line 79, in scatter_max
        out: Optional[torch.Tensor] = None,
        dim_size: Optional[int] = None) -> Tuple[torch.Tensor, torch.Tensor]:
    return torch.ops.torch_scatter.scatter_max(src, index, dim, out, dim_size)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
RuntimeError: "scatter" not implemented for 'Half'

Wansit99, 2024-04-26:
[my torch_scatter version is 2.0.6

> And when I update this files, I meet some problems
>
> File "/opt/data/private/whu_wsy/seg/FRNet-master-init/frnet/models/segmentors/frnet.py", line 83, in extract_feat voxel_dict = [...]

and my torch_scatter version is 2.0.6

Wansit99, 2024-04-26:
> And when I update this files, I meet some problems
> 
> File "/opt/data/private/whu_wsy/seg/FRNet-master-init/frnet/models/segmentors/frnet.py", line 83, in extract_feat voxel_dict = self.voxel_encoder(voxel_dict) File "/opt/conda/envs/frnet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl result = self.forward(*input, **kwargs) File "/opt/data/private/whu_wsy/seg/FRNet-master-init/frnet/models/voxel_encoders/frustum_encoder.py", line 106, in forward voxel_feats = torch_scatter.scatter_max( RuntimeError: The following operation failed in the TorchScript interpreter. Traceback of TorchScript (most recent call last): File "/opt/conda/envs/frnet/lib/python3.8/site-packages/torch_scatter/scatter.py", line 79, in scatter_max out: Optional[torch.Tensor] = None, dim_size: Optional[int] = None) -> Tuple[torch.Tensor, torch.Tensor]: return torch.ops.torch_scatter.scatter_max(src, index, dim, out, dim_size) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE RuntimeError: "scatter" not implemented for 'Half'
> 
> my torch_scatter version is 2.0.6

It works well in 2.0.8

Xiangxu-0103, 2024-04-26:
Thanks for your continuous attention to our work. In the updated version, we tried to use `iter-based` training instead of `epoch-based` training and use `mixed precision` (successful in `SemanticKITTI` while a little problem in `nuScenes`). All the models are trained with 4 GPUs and batch size is set to `4` for each GPU. If your configuration differs from ours, you may need to adjust the `max_iters` in the config.

Wansit99, 2024-04-26:
Thank you for your prompt response. After using the new config, I noticed a significant decrease in training speed, almost 1.5 times slower. Even after disabling mixed precision through onecycle-50k.py, there's still around a 1.3 times decrease in speed. What could be the reason for this?

Wansit99, 2024-04-26:
> Thanks for your continuous attention to our work. In the updated version, we tried to use `iter-based` training instead of `epoch-based` training and use `mixed precision` (successful in [...]

Could you clarify what nuScenes is? I've been experimenting with KITTI so far, but I'll need to work with nuScenes soon. Thank you!

Xiangxu-0103, 2024-04-27:
Here, we provide the [training log](https://github.com/Xiangxu-0103/FRNet/files/15136880/20240316_144229.log), which is trained with `4xb4`.

For nuScenes, you may refer to [website](https://www.nuscenes.org/) for downloading and prepare the dataset as mentioned in the document.

Wansit99, 2024-04-27:
> Here, we provide the [training log](https://github.com/Xiangxu-0103/FRNet/files/15136880/20240316_144229.log), which is trained with `4xb4`.
> 
> For nuScenes, you may refer to [website](https://www.nuscenes.org/) for downloading and prepare the dataset as mentioned in the document.

Thanks! But I have a question about the val batch_size, why it needs to be set to 1?

Wansit99, 2024-04-27:
> Here, we provide the [training log](https://github.com/Xiangxu-0103/FRNet/files/15136880/20240316_144229.log), which is trained with `4xb4`.
> 
> For nuScenes, you may refer to [website](https://www.nuscenes.org/) for downloading and prepare the dataset as mentioned in the document.

And I find the version in the INSTALL.md is not same as the log,could you update them? Thanks!

Wansit99, 2024-04-27:
> Here, we provide the [training log](https://github.com/Xiangxu-0103/FRNet/files/15136880/20240316_144229.log), which is trained with `4xb4`.
> 
> For nuScenes, you may refer to [website](https://www.nuscenes.org/) for downloading and prepare the dataset as mentioned in the document.

It works well when I update torch from 1.8.1 to 1.11.0
and this is the conda environment by conda list:

_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
absl-py                   2.1.0                    pypi_0    pypi
addict                    2.4.0                    pypi_0    pypi
aiofiles                  22.1.0                   pypi_0    pypi
aiosqlite                 0.20.0                   pypi_0    pypi
aliyun-python-sdk-core    2.15.1                   pypi_0    pypi
aliyun-python-sdk-kms     2.16.2                   pypi_0    pypi
anyio                     4.3.0                    pypi_0    pypi
argon2-cffi               23.1.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
arrow                     1.3.0                    pypi_0    pypi
asttokens                 2.4.1                    pypi_0    pypi
attrs                     23.2.0                   pypi_0    pypi
babel                     2.14.0                   pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
beautifulsoup4            4.12.3                   pypi_0    pypi
black                     24.4.2                   pypi_0    pypi
bleach                    6.1.0                    pypi_0    pypi
ca-certificates           2024.3.11            h06a4308_0  
cachetools                5.3.3                    pypi_0    pypi
certifi                   2024.2.2                 pypi_0    pypi
cffi                      1.16.0                   pypi_0    pypi
charset-normalizer        3.3.2                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
colorama                  0.4.6                    pypi_0    pypi
comm                      0.2.2                    pypi_0    pypi
contourpy                 1.1.1                    pypi_0    pypi
crcmod                    1.7                      pypi_0    pypi
cryptography              42.0.5                   pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
debugpy                   1.8.1                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
defusedxml                0.7.1                    pypi_0    pypi
deprecation               2.1.0                    pypi_0    pypi
descartes                 1.1.0                    pypi_0    pypi
entrypoints               0.4                      pypi_0    pypi
exceptiongroup            1.2.1                    pypi_0    pypi
executing                 2.0.1                    pypi_0    pypi
fastjsonschema            2.19.1                   pypi_0    pypi
fire                      0.6.0                    pypi_0    pypi
flake8                    7.0.0                    pypi_0    pypi
fonttools                 4.51.0                   pypi_0    pypi
fqdn                      1.5.1                    pypi_0    pypi
google-auth               2.29.0                   pypi_0    pypi
google-auth-oauthlib      1.0.0                    pypi_0    pypi
grpcio                    1.62.2                   pypi_0    pypi
idna                      3.7                      pypi_0    pypi
imageio                   2.34.1                   pypi_0    pypi
importlib-metadata        7.1.0                    pypi_0    pypi
importlib-resources       6.4.0                    pypi_0    pypi
iniconfig                 2.0.0                    pypi_0    pypi
ipykernel                 6.29.4                   pypi_0    pypi
ipython                   8.12.3                   pypi_0    pypi
ipython-genutils          0.2.0                    pypi_0    pypi
ipywidgets                8.1.2                    pypi_0    pypi
isoduration               20.11.0                  pypi_0    pypi
jedi                      0.19.1                   pypi_0    pypi
jinja2                    3.1.3                    pypi_0    pypi
jmespath                  0.10.0                   pypi_0    pypi
joblib                    1.4.0                    pypi_0    pypi
json5                     0.9.25                   pypi_0    pypi
jsonpointer               2.4                      pypi_0    pypi
[... 132 lines cut ...]
torch-scatter             2.0.9                    pypi_0    pypi
torchaudio                0.11.0+cu113             pypi_0    pypi
torchvision               0.12.0+cu113             pypi_0    pypi
tornado                   6.4                      pypi_0    pypi
tqdm                      4.65.2                   pypi_0    pypi
traitlets                 5.14.3                   pypi_0    pypi
trimesh                   4.3.1                    pypi_0    pypi
types-python-dateutil     2.9.0.20240316           pypi_0    pypi
typing-extensions         4.11.0                   pypi_0    pypi
tzdata                    2024.1                   pypi_0    pypi
uri-template              1.3.0                    pypi_0    pypi
urllib3                   1.26.18                  pypi_0    pypi
wcwidth                   0.2.13                   pypi_0    pypi
webcolors                 1.13                     pypi_0    pypi
[... cut ...]
wheel                     0.41.2           py38h06a4308_0  
widgetsnbextension        4.0.10                   pypi_0    pypi
xz                        5.4.6                h5eee18b_0  
y-py                      0.6.2                    pypi_0    pypi
yapf                      0.40.2                   pypi_0    pypi
ypy-websocket             0.8.4                    pypi_0    pypi
zipp                      3.18.1                   pypi_0    pypi
zlib                      1.2.13               h5eee18b_0

Wansit99, 2024-04-27:
Here is the mm version suitable for torch 1.11.0
'mmdet3d==1.4.0'

Xiangxu-0103, 2024-04-27:
I think the latest version is also compatible to our repo.

closed 2024-06-19 by Xiangxu-0103