Xiangxu-0103/FRNet #22: About export prediction results

Wansit99, 2024-05-09:
Could you please guide me on how to export prediction results using code? I want to submit them to the KITTI leaderboard.Thanks!

Xiangxu-0103, 2024-05-24:
Sorry for late reply. I suggest you customize a new `Evaluator` instead of `SegMetric`, in which you can define your own operation, such as remapping the predicted label to the original label and saving the label. It is easy to debug from my perspective.

Wansit99, 2024-05-24:
> Sorry for late reply. I suggest you customize a new `Evaluator` instead of `SegMetric`, in which you can define your own operation, such as remapping the predicted label to the original label and saving the label. It is easy to debug from my perspective.

Thanks!

closed 2024-06-19 by Xiangxu-0103