Xiangxu-0103/FRNet #24: Inference on different dataset

guidodumont, 2024-05-15:
Hi, I would like to use FRNet on the [UTCampus](https://amrl.cs.utexas.edu/coda/) dataset, however, I don't get the expected results. The code runs without any issues, but the results are terrible, see below. I use the provided config and checkpoints files corresponding to the KITTI dataset within the mmdetection3D framework.

In my opinion, the poor results can be caused by the difference in LiDAR sensors, the setup of the config files, or incompatibility in general. I already tried a couple of things in order to solve the issue, unfortunately, they all didn't work:
1. Tried the checkpoint and config file corresponding to the nuScenes dataset
2. Changed the fov_up and fov_down in both config files
3. Downsampled my point cloud to better match the resolution of KITTI and nuScenes

I hope that I gave you enough information, otherwise feel free to ask.

Kind regards,

Guido

[image]

ldkong1205, 2024-05-15:
Hi @guidodumont, can you show some examples of projecting your LiDAR point clouds to range view using the FOVs you tried? We cannot make a comment on your scenario until getting more information about your dataset. Thanks

guidodumont, 2024-06-20:
Sorry for the late response. The cause of the problem is probably the difference in sensor position (on top of a car/mobile robot) and the inconsistent intensity measurements. I have found an alternative method to obtain the data I need. I much appreciated your help!

closed 2024-06-20 by guidodumont