Xiangxu-0103/FRNet #29: fast-FRNet

xunkaixin, 2024-06-19:
can you provide the config and weights about fast-FRNet, thinks

Xiangxu-0103, 2024-06-19:
Hi @xunkaixin, It's easy to implement Fast-FRNet with existing config, as the 2D representation is set to `(32, 360)` and the backbone for Frustum is `ResNet-18`.

closed 2024-06-19 by xunkaixin

RhythmOfTheRain-Byte, 2024-11-27:
> Hi @xunkaixin, It's easy to implement Fast-FRNet with existing config, as the shape of 2D representation is `(32, 360)` and the backbone for Frustum is `ResNet-18`.

Excuse me, I would like to ask how to implement Fast FRNet based on the existing config. What modifications should I make and what corresponding changes are required? 
I apologize for disturbing you as I'm not very familiar with this framework. Thank you!