Xiangxu-0103/FRNet #31: add some points, original point's label will be change

xunkaixin, 2024-06-25:
Original point cloud p0, p1, p2, p3, ... pn,  Obtain inference results L0, L1, L2, L3,... Ln
Add some points (from original) to the original point cloud to form a new point cloud p0, p1, p2, p3, ... pn, p2, p3,  Will cause a change in the inference category of the original point cloud

Is it my mistake or is it that the result of  algorithm  is like this？

Xiangxu-0103, 2024-06-25:
I am sorry that I cannot understand what you mean. Can you provide your questions more clearly?

xunkaixin, 2024-06-25:
Assuming there are 1000 points in the original point cloud, infer and obtain the category of the points; Randomly select 10 points from the original point cloud and add them to the end of the original point cloud to form 1010 points, and then perform inference. The inference result will change the inference result of 1000 points; In other words, 10 additional points changed the category of 1000 points

closed 2024-07-02 by xunkaixin