Xiangxu-0103/FRNet #41: about labels_map

kyeonghyeon0314, 2025-03-19:
In ```labels_map```, I see that ```bus``` is mapped to ```other-vehicle```. 
However, why is  ```mmdetection3d\mmdet3d\datasets\semantickitti_dataset.py```'s METAINFO stated as ```bus``` instead of ```other-vehicle```?

Please check if it is correct to change it like below.

```
    METAINFO = {
        'classes': ('car', 'bicycle', 'motorcycle', 'truck', 'other-vehicle', 'person',
                    'bicyclist', 'motorcyclist', 'road', 'parking', 'sidewalk',
                    'other-ground', 'building', 'fence', 'vegetation',
                    'trunk', 'terrian', 'pole', 'traffic-sign'),
        'palette': [[100, 150, 245], [100, 230, 245], [30, 60, 150],
                    [80, 30, 180], [0, 0, 255], [255, 30, 30],
                    [255, 40, 200], [150, 30, 90], [255, 0, 255],
                    [255, 150, 255], [75, 0, 75], [175, 0, 75], [255, 200, 0],
                    [255, 120, 50], [0, 175, 0], [135, 60, 0], [150, 240, 80],
                    [255, 240, 150], [255, 0, 0] ],
        'seg_valid_class_ids':
        tuple(range(19)),
        'seg_all_class_ids':
        tuple(range(19)),
    }
```

Xiangxu-0103, 2025-03-19:
The `labels_map` is set following the official script. More details about SemanticKITTI can be referred to [this file](https://github.com/PRBonn/semantic-kitti-api/blob/master/config/semantic-kitti.yaml).

kyeonghyeon0314, 2025-03-19:
Thank you for your quick reply.
Additionally, how do I solve the **0.19%** ***IoU*** of ```motorcyclist``` class if I proceed with the provided ```pre-trained checkpoints```?

Xiangxu-0103, 2025-03-19:
`motorcyclist` achieves nearly 0% IoU due to the class imbalance and the lack of sufficient points which are labeled as `motorcyclist` in the validation set. It seems that current methods have not well addressed this. However, in the test set, `motorcyclist` can achieve 57.3% IoU.

kyeonghyeon0314, 2025-03-19:
Thanks to your explanation, the curiosity was solved. Thank you.

closed 2025-03-19 by kyeonghyeon0314